micropython-IMU/micropython-fusion #1: pyb library?

kswain1, 2015-11-13:
I am having a problem importing the pub library where I can I find it. Tried using pip, and didn't have any luck.

kswain1, 2015-11-13:
I'm using python 2.7, and I'm not using a micro python board. I am using OS X YOSEMITE to filter information.

kswain1, 2015-11-13:
I only need the pyb module to complete this method Integrate to yield quaternion. Is there any other way I can do the integration with another module besides pyb?

turbinenreiter, 2015-11-13:
You will have to replace the pyb calls with calls to time: https://docs.python.org/2/library/time.html

kswain1, 2015-11-13:
Yes I have added the time function to instead of pyb.
Do you know if this is similar function to pyb micros

import time 
def micros():
     return time.time()/1000000

def elapsed_micros(start_time):
      return time.time()/1000000 - start_time

If so, I am currently getting all zero's for my yaw, pitch, and roll,which is definitely wrong. 
What are some average angles to expect?

peterhinch, 2015-11-13:
time.time() returns seconds so you need to multiply by 1000000 to get microseconds rather than divide.

kswain1, 2015-11-13:
Thank you will try it in the next couple of hours to let you know the results, so we can close this issue.

neep8, 2015-11-14:
Hi I won't have my computer until the end of the month... But I was looking at getting the IMU code working in python3 with a mpu6050. Like you I struggled to find the pyb library. Hope it works for you.

kswain1, 2015-11-14:
Hey I believed the solutions works! I am getting results, but my next question is for the calibration function during the initial time does the algorithm start the position at (0,0,0) until the chip is moved? If so I am having problems with the calibration portion of the algorithm.

turbinenreiter, 2015-11-15:
Make sure to share your patches.

kswain1, 2015-11-16:
I believe everything is working. Patch Update 
import time 
def micros():
return time.time()*1000000

def elapsed_micros(start_time):
return time.time()*1000000 - start_time

replace these functions with pyb, and you should be ok.

closed 2015-11-16 by kswain1