micropython-IMU/micropython-fusion #2: Beta Setting

BenMosbah, 2015-12-21:
Hello, I don't understand why beta is set to be  sqrt(3/4)*GyroMeasError
In the original study, beta is indeed set this way but there is no explanation.

I read that beta is similar to I in a PID control sense, I don't really understand why, could someone please explain?

Thanks in advance

peterhinch, 2015-12-22:
Alas I'm no expert on the maths and copied the original setting. If you find an informative URL I'd be interested to see it and I'd include it in the docs.

BenMosbah, 2015-12-22:
I think that it is a statistics' formula used to estimate the error introduced by a constant bias (GyroMeasError), but I'm no expert in math either, so I'm not sure.

I see that you don't define "zeta", is it on purpose that you do not account for gyro bias drit as recommended by Madgwick's paper?

Another question, how did you figure out the formulas giving heading, pitch and roll from quaternion coordinates ? (I reckon these are not the same formulas provided in Madgwick's paper)

Thanks in advance

peterhinch, 2015-12-23:
I was unsure about the issue of gyro bias drift. I suspect that modern MEMS sensors are probably better in that respect than traditional spinning disks. So I left the issue of "zeta" unresolved. I evaluated the results with a MicroPython board running an MPU9150 sensor and driving an LCD display giving heading, pitch and roll figures. The results seem consistent (after magnetometer calibration) although I can't claim to have done precise measurements.

I ported the code from a C solution, studying the Madgwick paper for reference. The code includes references to the original source 
https://github.com/kriswiner/MPU-9250/blob/master/quaternionFilters.ino
If there are errors in the source it is almost certain that they have propagated into my port.

BenMosbah, 2015-12-27:
Hello,

I made some research and eventually asked x-io.
It seems that the sqrt(3/4) factor comes from the definition of beta as the norm of a quaternion's derivative (hence the sqrt!).
This can be seen on Page 13 of the internal report 
http://www.x-io.co.uk/res/doc/madgwick_internal_report.pdf

Nevertheless, I have a question I hope you can answer. 
Upon reading the code it seems that beta is there to correct the gyro bias (ergo the value it gives when it is not moving) but this kind of error is not mean zero, is it?

What bothers me is that in the report it is said that beta "represents all mean zero gyroscope measurement errors"  (while zeta is there for mean zero errors)

peterhinch, 2015-12-28:
I don't think I can offer any further enlightenment on this apart from pointing out that in Madgwick's testing he set zeta to zero (page 18).