micropython-IMU/micropython-fusion #10: Cannot get heading to make sense

dannystaple, 2020-08-18:
I have an ICM 20948 IMU (another Invensense) on a Raspberry Pi.

I can make the accelerometer and gyro fusion work well. The tilt and roll make sense when using the update_nomag, or when updating with the mag.

I have been able to calibrate the magnetometer for hard iron offsets - using scatter plot convergence similar to https://learn.adafruit.com/adafruit-sensorlab-magnetometer-calibration/calibration-with-raspberry-pi-using-blinka.

My problem is that heading reads a value of -32 for the heading when facing North, -81 facing East. -62 facing south and -85 facing west.

Have you got any ideas?

peterhinch, 2020-08-19:
Hard for me to comment as I have no experience of the device or its driver. The author of the device driver might have some ideas. Otherwise I would ask in [the forum](http://forum.micropython.org/index.php).

In general calibrating magnetometers requires care. The approach I've found best is to rotate the device slowly around each orthogonal axis - hopefully the driver has the capability of processing and storing calibration data as a quick glance at the datasheet suggests that the hardware doesn't have this capability.

dannystaple, 2020-09-07:
I do have one possible lead. The Datasheet for the device shows that the direction of the Y and Z access are opposite (negative) to those of the gyroscope and accelerometer. I’ll try account for that and see if it helps. 
I’ve been storing the calibration data locally, and taking it off the device in my code. 

Also - it feels like an almighty hack, but if I discard the z-component completely, the magnetometer produces a vector that starts to make sense -> pointing vaguely north.
Danny

dannystaple, 2020-12-01:
Going to close this - I couldn't make this code work, but I think it was a bit of a misunderstanding of the interaction of the coordinate systems. Perhaps I will try again later to integrate this code.

closed 2020-12-01 by dannystaple

gretel, 2020-12-01:
my `mpu9250` just required to get calibrated and have the resulting data stored and used on each boot to achieve reproducible precision, especially on the yaw axis. please also mind the magnetic vs "true" north issue. 

in regards to my prototype - documentation is not done but the code is working fine - https://github.com/gretel/dogspeed