micropython-IMU/micropython-fusion #11: How to get position updates?

OlegJakushkin, 2020-11-09:
So I see how we get heading, pitch, roll. How to get position updates?

peterhinch, 2020-11-09:
That is not possible using an IMU. In theory it can be done by double integrating the accelerometer, but in practice errors accumulate exponentially. I believe it can be done with extreme precision accelerometers, but not with ones we can buy.

You need to find another way: counting wheel revolutions on a wheeled robot, or using GPS.

OlegJakushkin, 2020-11-09:
[These guys ](https://github.com/thomas-haslwanter/scikit-kinematics/blob/master/skinematics/imus.py#L149-L151) tend to promise position+quaternion pairs. Here is their[ implementation sample](https://github.com/thomas-haslwanter/scikit-kinematics/blob/c0d0c93c5e75a8f07d243d904ea7baf6f0ee64b7/skinematics/imus.py#L329), their license is also MITish.

peterhinch, 2020-11-09:
I would be fascinated to hear of any success with this. My own experiments produced exactly the (bad) result I expected.

As far as I can see, basic physics requires double integration with its inherent vice of amplifying errors.  [This code comment](https://github.com/thomas-haslwanter/scikit-kinematics/blob/c0d0c93c5e75a8f07d243d904ea7baf6f0ee64b7/skinematics/imus.py#L442) seems to suggest that they are double integrating. However I'm willing to be convinced that they have discovered something I've missed. Perhaps there is some clever filtering technique?

claudiorodr, 2020-12-20:
@OlegJakushkin have you had any success from that implementation? As @peterhinch mentioned, I also haven't any good results double integrating due to cumulative errors.