micropython-IMU/micropython-fusion #14: MPU6050 Heading always zero

fanlessfan, 2021-03-03:
Hi @peterhinch 

Thank you for your help on MPU6050. It works now, but the Heading is always 0. is it normal for 6050?

thx

Update time (uS): 1041
Heading, Pitch, Roll:   0.000  -0.011  -0.004
Heading, Pitch, Roll:   0.000  -5.174  -3.649
Heading, Pitch, Roll:   0.000  -5.389  -2.731
Heading, Pitch, Roll:   0.000 -29.774  -9.453
Heading, Pitch, Roll:   0.000  -5.891  -1.644

peterhinch, 2021-03-04:
Yes. As the docs make clear, the MPU6050 has no magnetometer so deducing heading is impossible. You need an MPU9150 or MPU9250.

fanlessfan, 2021-03-04:
Thank you @peterhinch for the clarification

peterhinch, 2021-03-04:
Please could you read the docs: I put considerable effort into explaining these details, with special emphasis on the practical details around magnetometers. All the queries you have raised are covered in the docs.

closed 2021-03-04 by peterhinch