micropython-IMU/micropython-fusion #15: what is switch in fusiontest.py

fanlessfan, 2021-03-03:
Hi @peterhinch 

when I try to run the fusion test.py for MPU9250, there is code for external switch. Could you explain what that is and how to use it? I am using ESP32 board.

thx

peterhinch, 2021-03-04:
See [this line](https://github.com/micropython-IMU/micropython-fusion/blob/43748e9cc277ed7d17445025f48b774e2cc2288e/fusiontest.py#L19). It allows for a physical switch or pushbutton connected between a pin and Gnd. This enables you to tell the program when calibration is done. However the purpose of calibration is to correct the magnetometer, and your device doesn't have one.

closed 2021-03-04 by peterhinch

fanlessfan, 2021-03-04:
Hi @peterhinch , I did see that line. Could you explain more about the calibration process? I do have MPU9250 try.

thx

peterhinch, 2021-03-04:
Please read the docs. If there is a specific point in the docs which you don't understand, please point me to the paragraph which is unclear. That way I can improve the docs for others.