micropython-IMU/micropython-fusion #17: Python implementation on Raspberry Pi - Fused values statics

alcor55, 2022-01-20:
Hello!
I have just included the library in a python project, in a RaspberryPi 0 2w, with python3.7 and an ICM20948 9DOF IMU as it follows:

`
from icm20948 import ICM20948
import time
import gc
from fusion import Fusion

def time_diff(start, end):
    return (start - end)/1000000

fuse = Fusion(time_diff)

count = 0
while True:
    x, y, z = self.imu.read_magnetometer_data()
    ax, ay, az, gx, gy, gz = self.imu.read_accelerometer_gyro_data()
    ts = time.time()
    self.fuse.update([ax, ay, az], [gx, gy, gz], [x, y, z], ts) # Note blocking mag read
    if count % 25 == 0:
        print("{:7.3f} {:7.3f} {:7.3f} - {:05.2f}|{:05.2f}|{:05.2f}|{:05.2f}|{:05.2f}|{:05.2f}|{:05.2f}|{:05.2f}|{:05.2f}".format(
        self.fuse.heading, self.fuse.pitch, self.fuse.roll, ax, ay, az, gx, gy, gz, x, y, z))
        count += 1
`

It seems to work errorless but the fused values (heading, pitch and roll) are either static or either the variation is negligible:

`
heading  pitch       roll.         ax     ay       az.     gx      gy     gz          x       y        z
 -0.002   0.005   0.005 - 00.07|00.00|01.04|00.05|00.21|00.27|-12.30|35.10|94.65
 -0.002   0.005   0.005 - 00.07|00.00|01.04|00.10|00.42|00.13|-12.90|33.90|96.00
 -0.002   0.005   0.005 - 00.07|00.00|01.04|-0.05|00.26|00.26|-12.90|35.70|96.00
 -0.002   0.005   0.005 - 00.07|00.01|01.04|00.22|00.33|00.40|-11.85|35.70|94.95
 -0.002   0.005   0.005 - 00.07|00.01|01.04|00.24|00.40|00.30|-13.35|36.15|96.00
 -0.002   0.005   0.005 - 00.15|-0.25|01.01|-78.54|18.65|-5.38|-16.20|26.40|95.55
 -0.002   0.005   0.005 - 00.09|-0.65|00.76|-3.21|03.21|-7.35|-13.35|09.75|87.00
 -0.002   0.005   0.005 - 00.16|-0.77|00.64|-8.72|-9.74|-2.31|-18.15|02.55|83.85
 -0.002   0.005   0.005 - -0.06|00.24|00.99|110.12|-7.94|23.41|-9.75|48.60|88.80
 -0.002   0.005   0.005 - -0.17|00.35|00.98|-91.23|-17.25|-118.19|-4.50|49.65|84.60
 -0.002   0.005   0.005 - 00.50|-0.52|00.94|-41.48|-78.02|-17.75|-44.70|-6.30|80.55
 -0.002   0.005   0.005 - 00.54|-0.61|00.61|149.80|40.49|208.66|-37.35|-11.10|75.30
 -0.002   0.005   0.005 - 00.19|00.33|00.92|01.63|-1.18|-124.72|-16.05|59.55|90.45
 -0.002   0.005   0.005 - 00.11|00.16|00.94|-4.61|-2.31|09.70|-10.35|49.05|90.30
 -0.002   0.005   0.005 - 00.09|00.09|01.05|-24.18|05.10|-32.29|-13.35|47.10|95.25
 -0.002   0.005   0.005 - 00.06|-0.12|00.98|-0.69|-16.33|-36.27|-12.30|20.70|91.50
 -0.002   0.005   0.005 - 00.31|-0.15|01.00|-4.06|-1.60|-28.98|-32.55|01.05|91.20
 -0.002   0.005   0.005 - 00.12|-0.11|00.94|24.14|10.39|88.27|-15.45|21.15|93.60
 -0.002   0.005   0.005 - -0.11|00.71|00.62|42.82|13.69|10.38|-9.75|66.15|72.75
 -0.002   0.005   0.005 - -0.18|00.63|00.85|-108.66|-38.09|-25.57|-7.95|63.90|74.85
 -0.002   0.005   0.005 - 00.21|-0.79|00.60|-49.85|-16.36|-5.11|-17.85|05.85|85.95
 -0.002   0.005   0.005 - -0.48|-0.48|00.77|75.34|91.73|75.86|-1.50|18.45|71.55
 -0.002   0.006   0.005 - -0.89|-0.06|00.61|-16.23|-110.88|-1.24|03.15|35.70|54.00
 -0.002   0.006   0.005 - 00.34|00.13|00.95|20.72|-48.60|12.53|-21.15|45.75|97.20
 -0.002   0.006   0.005 - 00.16|00.98|00.32|112.11|84.21|-46.33|-11.85|69.45|66.60
 -0.002   0.006   0.005 - 00.02|-1.00|00.92|-23.82|-179.43|-74.77|-14.40|03.60|81.45
 -0.002   0.006   0.005 - -0.54|00.90|00.02|-8.20|62.37|-83.08|-4.50|58.05|46.80
 -0.002   0.006   0.005 - 00.21|-0.54|01.05|158.70|-11.55|25.06|-19.50|21.00|96.90
 -0.002   0.006   0.005 - -0.13|-0.40|01.02|-95.34|-34.41|-24.75|-6.30|25.95|87.00
 -0.002   0.006   0.005 - 00.12|00.09|01.00|11.50|-1.62|-9.82|-15.00|41.25|95.25
 -0.002   0.006   0.005 - 00.09|00.14|01.04|12.60|05.79|12.21|-12.00|45.15|91.05
 -0.002   0.006   0.005 - -0.02|00.36|00.97|29.73|14.14|-9.98|-10.95|42.15|93.75
 -0.002   0.006   0.005 - 00.12|-0.02|00.96|02.65|10.74|-18.39|-12.45|42.15|94.50
 -0.002   0.006   0.005 - 00.03|00.08|01.09|02.56|06.44|-2.18|-9.45|42.00|92.55
 -0.002   0.006   0.005 - 00.05|00.05|01.03|-17.07|-4.51|02.40|-13.65|42.00|93.60
 -0.002   0.006   0.005 - 00.11|00.00|01.01|-0.56|-1.44|-1.69|-12.90|40.95|94.95
 -0.003   0.006   0.005 - -0.10|00.62|01.14|-9.95|-59.97|-107.58|-12.30|39.75|96.00
 -0.003   0.006   0.005 - -0.87|-0.12|01.11|-19.57|-140.28|12.60|-13.65|39.00|95.25
 -0.003   0.006   0.005 - -0.12|00.00|00.46|-17.89|62.86|-23.93|-14.10|41.40|93.00
 -0.003   0.006   0.005 - 00.78|00.04|01.04|-82.00|93.40|-250.14|-15.00|35.10|96.00
 -0.003   0.006   0.006 - 00.45|00.27|01.00|-8.88|35.37|71.79|-18.00|48.30|96.60
 -0.003   0.006   0.006 - 00.09|00.05|01.06|04.44|25.15|-34.18|-14.10|42.15|96.60
 -0.003   0.006   0.006 - -0.07|00.85|00.59|13.27|-0.63|-1.90|-10.05|66.90|71.40
 -0.003   0.006   0.006 - 00.18|-0.84|00.68|65.41|03.37|13.23|-19.05|03.45|85.80
 -0.003   0.006   0.006 - 00.09|00.04|01.03|-1.47|-14.40|-0.32|-10.80|39.00|94.65
 -0.003   0.006   0.006 - 00.06|00.00|01.04|00.35|00.21|-0.03|-12.30|39.45|96.75
 -0.003   0.006   0.006 - 00.06|00.01|01.04|00.15|-0.03|00.34|-12.90|39.00|94.65
 -0.003   0.006   0.006 - 00.07|00.00|01.06|-0.11|00.13|00.38|-11.40|39.60|94.50
`

Can you help me?

alcor55, 2022-01-22:
Solved. I set the time-stamp with time.time_ns() and now it works properly.
I hope this solution helps somebody else.

peterhinch, 2022-01-22:
Thanks for the report.

closed 2022-01-22 by peterhinch